marcelstoer/nodemcu-custom-build #8: Include more details in email?

nfriedly, 2015-07-30:
Hey, would it be possible to include the list of modules (and fonts!) in the email that gets sent out for each build?

marcelstoer, 2015-07-30:
While the list of modules is already included in the emails you got today at 20:19, 20:57, and 21:08 the list of fonts isn't. Since most people don't use the u8g module (reference: http://frightanic.com/nodemcu-custom-build/stats.php) the list of fonts would be of no use to them.

Even though I won't be adding the font list to the email any time soon I keep this issue open.

nfriedly, 2015-07-30:
Oh, doh, it was hidden behind Gmail's "show quoted text".

And, yea, the font list only makes sense when u8g is included.

marcelstoer, 2015-08-01:
> it was hidden behind Gmail's "show quoted text"

Would be interesting to know what exactly makes Google think that'd be quoted text.

nfriedly, 2015-08-01:
Well, it probably wasn't hidden in the first one, but I did 3 builds in a
row and selected the same options each time. Then later I wanted to figure
out which fonts I had selected and if I had included some module...

marcelstoer, 2016-07-31:
We're now fast approaching build number 50'000 and this request has never come up again -> closing.

closed 2016-07-31 by marcelstoer