marcelstoer/nodemcu-custom-build #9: remember selected options between builds

nfriedly, 2015-07-30:
I'd appreciate it if the site remembered what modules and fonts I had selected the previous time (by saving it in a cookie or localStorage or whatever)

I'd be happy to send a PR for this one, but I don't see the html files anywhere (?)

marcelstoer, 2015-07-30:
Even though this is something I had initially on my list of must-have features it took more than 1100 unique users (who commissioned nearly 2000 builds) until it was first requested by a user :astonished:. Not sure what I should make this...

Thanks for the PR offer but the PHP files are closed source.

nfriedly, 2015-08-03:
Yea, I think most people select their modules and then they're done with it for a while, (or else they come back wanting different modules, so remembering the previous selection doesn't matter.)

This got me because I wanted to try out a number of different fonts to see how they looked on my display, but with a fairly low limit, I ended up doing several builds in a row with the same modules (and some of the same fonts) selected.

marcelstoer, 2015-08-03:
To commission several builds in quick succession it usually works fine if you just use "browser back" on the confirmation page. Most browsers should keep the form values.

marcelstoer, 2016-07-31:
We're now fast approaching build number 50'000 and this request has never come up again -> closing.

closed 2016-07-31 by marcelstoer