marcelstoer/nodemcu-custom-build #11: Option to specify flash size

Aeprox, 2015-10-25:
Since updating to SDK 1.4.0, ESP's with 4Mbit of flash are hitting the limits of their flash size. When building with #define FLASH_AUTOSIZE, the size of a build with SSL enabled and just the basic modules is too big for a 512K ESP's. Flashing these images to the ESP causes it to bootloop or other strange behavior.
However, when specifying #define FLASH_512K in user_config.h, the compiler gives an error indicating that it won't fit into irom.text. 
Would it therefor be possible to include the flash-size option in the build? This prevents confused users in the future who flashed a succeeded build that somehow still causes bootloops.

marcelstoer, 2015-10-25:
I see the problem i.e. I was aware of that.

> Would it therefor be possible to include the flash-size option in the build?

It's certainly possible but IMO it isn't good idea. When the build fails the user gets a notification email like "Sorry, your NodeMCU custom build failed. The maintainer of this site was notified to take a look. Stay tuned."
_I_, on the other hand, get an email with all the details of the failure from Travis CI. I then analyze the problem, take appropriate measures and get in touch with the user. I guess you see why letting the build fail wouldn't be a good move for me.

How about adding a note in the branches section?

Aeprox, 2015-10-25:
Yeah, I see the problem with that. A note would work but it's not exactly fool-proof. People are lazy, reading is hard etc :)
(I'm currently not even getting that warning anymore myself, so it's hard to think of a way to implement this. How/when and where is the value in user_config.h accessed by the compiling process? Couldn't we parse that steps output and act upon that?)

marcelstoer, 2015-10-25:
> A note would work but it's not exactly fool-proof.

I'll add that later today for starters. Ultimately what I'd like to have is an internal calculator that can predict how much memory a given branch/modules/options combination is going to consume.

> How/when and where is the value in user_config.h accessed by the compiling process?

https://github.com/nodemcu/nodemcu-firmware/blob/master/app/platform/cpu_esp8266.h#L20 defines this and then uses it to calculate other constants which are then used in the code to address memory segments.

> Couldn't we parse that steps output and act upon that?

No, not really. The email I get from Travis CI is a JSON object(!) with all [sorts of info](http://docs.travis-ci.com/user/notifications/#Webhooks-Delivery-Format) but not the log itself. The log will state something like "'.text' will not fit in region `iram1_0_seg`" (or `irom.text` as you said). So, the term "will not fit in region" _is_ something to look out for but since the CI system is external options are very limited. If I had more time to dig into this there may be alternatives. However, since the service is quite popular (>400 builds per week) I need this to be 100% automatic.

closed 2016-12-28 by marcelstoer