marcelstoer/nodemcu-custom-build #14: Add bit rate selection

karrots, 2015-11-08:
Commit nodemcu/nodemcu-firmware@66e65bd3c185703964075b2d3bda5eb1ffe12dc1 added support for custom baud rates in the user_config.h. It would be great to see this functionality on the site. I realise it may not be wise to add it until the master branch has the same patch after things are merged.

marcelstoer, 2016-01-24:
As issue 872 in the NodeMCU repo will provide "sane" defaults for this I'm not sure it'll still be relevant to define your own. Agree?

karrots, 2016-01-25:
Yea, all i really wanted was 115,200 so assuming that goes through it seems reasonable.

marcelstoer, 2016-03-09:
https://github.com/nodemcu/nodemcu-firmware/pull/1127 will soon land -> won't fix this one.

closed 2016-03-09 by marcelstoer