marcelstoer/nodemcu-custom-build #21: Arbitrary repository as source for build

devsaurus, 2016-07-31:
As a developer, I would benefit from a feature that allows users to build from an arbitrary fork/branch.

I had a couple of support cases where users flagged bugs or feature requests but weren't able to verify the result since git/branch and local compile was a too high barrier for them. My approach in such cases is to provide a pre-built image for verification. It would be simpler to just tell them to build from the specific branch.

marcelstoer, 2016-07-31:
I know...it'd be nice to have, indeed. On the other hand if a developer isn't willing to use the Docker image, which is close to being a no-brainer IMO, so we can _help him/her_ I'd personally stop right there.
The problem with building from arbitrary forks/branches is the modules list which is static and not derived from `user_modules.h`. There are reasons for that and ATM I don't want to change that. However, I can imagine a bare-bone mode without any GUI options which would require that you edit all config files manually and commit & push them to your repo.

devsaurus, 2016-09-03:
Ok, docker seems to be a feasible alternative.

closed 2016-09-03 by devsaurus