marcelstoer/nodemcu-custom-build #25: In-progress builds canceled by new build request

stephenwhittle, 2017-11-01:
I've attempted to request a custom build a few times over the last day or so and didn't receive an email for any of them, so I found the link to the travis output and looked at the build history - it looks like builds that are in progress are being arbitrarily canceled by new build requests that arrive during the process. 
The message in the build log is : 

> This job was cancelled because the "Auto Cancellation" feature is currently enabled, and a more recent build (#160648) for branch builds came in while this job was waiting to be processed.

Would it be possible to either disable this feature or set Travis to send the email anyway, so we can at least see that the build's been canceled and need to try again? 
Thanks.

marcelstoer, 2017-11-01:
Thanks! You're the first to report this. Never heard of "Auto Cancellation", Travis CI must have introduced that just recently.

Currently looking into this.

marcelstoer, 2017-11-01:
> I found the link to the travis output

Where was that?

Just found https://blog.travis-ci.com/2017-09-21-default-auto-cancellation and turned off this feature. I think Travis made a mistake turning it _on_ by default w/o notifying owners of affected build jobs.

closed 2017-11-01 by marcelstoer

stephenwhittle, 2017-11-01:
So I found the travis output from the link posted in https://github.com/marcelstoer/nodemcu-custom-build/issues/23 and just looked at the 'build history' after re-running a build request.
Thanks for prompt attention to the issue - I'll run a build now and see if it works.

marcelstoer, 2017-11-02:
I restarted all cancelled builds going as far back as 20h (from when I wrote the previous comment). They all finished successfully.