marcelstoer/nodemcu-custom-build #26: MQTT

chathurangawijetunge, 2018-02-04:
hi i am using the code

m = mqtt.Client(node.chipid(), 5, "", "")
m:lwt("/lwt", "offline", 0, 0)
m:on("connect", function(client) print ("connected") end)
m:on("offline", function(client) print ("offline") end)

-- on publish message receive event
m:on("message", function(client, topic, data)
print("incomming...",topic .. ":" )
if data ~= nil then
print(data)
end
end)

function connect()
print("connecting.....")
m:connect("broker.hivemq.com", 1883, 0, function(client)
print("connected")

client:subscribe("/sw1", 0, function(client) print("subscribe success") end)
client:publish("/sw1", "hello", 0, 0, function(client) print("sent") end)
end,
function(client, reason)
print("failed reason: " .. reason)
tmr.alarm(6,2000, 0, function() connect() end)
end)
end

connect()
this works fine...
until i remove the phone line from my rougher and re connects after some time this code re connects to the MQTT broker.
but after 5 seconds it fires the offline function and reconnects, this happens every 5 seconds there after...
is it a bug????

chathurangawijetunge, 2018-02-04:
Same issue in #2245
Pls advise how to overcome this problem.

marcelstoer, 2018-02-04:
This is not the right place to address firmware problems.
> Pls advise how to overcome this problem.

Someone should fix https://github.com/nodemcu/nodemcu-firmware/issues/2197.

closed 2018-02-04 by marcelstoer