marcelstoer/nodemcu-custom-build #32: Allow copy/paste emails

RenaKunisaki, 2018-11-21:
It's 2018, our browsers automatically fill in the email field. Being forced to copy it by hand is only going to cause *more* typos.

mk-pmb, 2019-12-31:
Please have a lookt at `build.config`, line 1:

> email=suvft@rkpvgr.pbz

and the occurrences of `X_EMAIL` in `.travis.yml` for how to build without having to re-enter your mail address.
I agree that the Web UI could be a lot cooler, but I think it's good enough for one-shot builds. For to-be-repeated builds, I'm working on [a github action](https://github.com/mk-pmb/nodemcu-build-as-github-action) to build them.

marcelstoer, 2020-01-01:
> I'm working on a github action to build them.

I'm curious to see how that turns out. Go for it! 👍 

> For to-be-repeated builds...

... all you need are probably two to cURL statements (never tried). Just sayin' 😜

mk-pmb, 2020-01-02:
It seems to work now. At least it builds something and provides it as artifact for download. Anyone who already has custom build config header files, and an expectation of what the firmware download should be or do, please fork, rename and try with your configs.

https://github.com/mk-pmb/nodemcu-firmware-daily-vanilla/

marcelstoer, 2020-02-17:
The repeat-email field is history.

closed 2020-02-17 by marcelstoer