marcelstoer/nodemcu-custom-build #33: Changes to be compatable with SDK 3.0 builds

TerryE, 2019-04-07:
With SDK 3.0, the size and location of the SPIFFS and LFS partitions are set in the Partition Table at image flashing and not in the compile.  The following defines are now in the **Here areafter, there be demons** section and should not be edited.
```C
#define LUA_FLASH_STORE                   0x0
#define SPIFFS_FIXED_LOCATION             0x0
#define SPIFFS_MAX_FILESYSTEM_SIZE        (~0x0)
```
The `set-config.sh` comments these out by prefixing `//` which causes a compile error during the make

TerryE, 2019-04-08:
In set-config.sh, we need to remove the lines 7-9, 25-40 and 58-60, moving the sed command down to 61.  The wrapper web app needs similarly updated.

marcelstoer, 2019-04-08:
> we need to remove

_Conditionally_ "remove" as any change has to be backwards compatible. The same scripts are used for `master` (and the frozen 1.5 branch).

TerryE, 2019-04-08:
> Conditionally "remove" 

Agreed.  We need an extra `build.config` parameter `SDK-3=true` to bypass this.

marcelstoer, 2019-04-08:
Oh no, these scripts need to be totally self-contained and figure out how to behave themselves (e.g. u8g vs. u8g2 https://github.com/marcelstoer/nodemcu-custom-build/blob/master/set-displays.sh#L35).

TerryE, 2019-04-08:
The algorithm is different for pre-PT and PT versions introduced with SDK 3.0. How can they figure this out themselves?  Surely this fact needs to be passed as a context?

marcelstoer, 2019-04-08:
Checking for the existence of `NODEMCU_EAGLEROM_PARTITION` in `user_config.h`? Or checking for `tools/nodemcu-partition.py`?

TerryE, 2019-04-08:
:+1: That would work :smile:

marcelstoer, 2019-04-19:
Fixed in df0112261ed255af4d546f4bbcb124124358669f

closed 2019-04-19 by marcelstoer

TerryE, 2019-04-19:
But unfortunately this will need tweaking again.  From feedback, I've decided that we should at least honour the `user_config.h` settings for the LFS and SPIFFS sizes as a default.  I will give you the fix when I issue the PR.

TerryE, 2019-05-01:
@marcel, sorry but following developer feedback, I have decided to retain the option to set the LFS and SPIFFS sizes in the `user_config.h`.   If you look at the proposed [user_config.h](/TerryE/nodemcu-firmware/blob/sdk-3.0-follow-up/app/include/user_config.h),  Then you will see that I have reinstated:
```C
//#define LUA_FLASH_STORE                   0x10000
... 
//#define SPIFFS_MAX_FILESYSTEM_SIZE 0x20000
```
There is a later `#ifdef` guard on these at line 220:227, but I have added a couple of syntactically ignored whitespaces `#  define ...` so that we can now drop the test at set-config.sh:L18; drop the # char in Lines 2, 34 and 39 and change the patterns at L42:44 to ignore the `#  define` cases:
```
sed -e "s!^.*\\(#define *LUA_FLASH_STORE\\).*!$lfs!" \
    -e "s!^.*\\(#define *SPIFFS_FIXED_LOCATION\\).*!$spiffs_base!" \
    -e "s!^.*\\(#define *SPIFFS_MAX_FILESYSTEM_SIZE\\).*!$spiffs_size!" \
```
plus reintroducing the size options in the online tool. Only the SPIFFS fixed location option is dropped, but so long we you drop this from the online choice the above changes should work.

I'll do the firmware merge later this evening.