marcelstoer/nodemcu-custom-build #35: No LFS partition

danjohn, 2019-09-18:
The web builder seems to not respect the LFS settings:

I tried to web-build a firmware with 64k, 96k and 128k LFS, but ended up with images, that claim to have no LFS partition (as node.flashreload() states; node.info("build_config").lfs_size = 0).

An old image web-built on 2019-08-14 works (has a LFS partition). Something obviousely changed (with the switch to NodeMCU 3.0 maybe?)...

danjohn, 2019-09-18:
Nice bash script ;). I'm not sure how I can help with this issue.

To be sure, we are talking about the same thing:
I committed a build on <https://nodemcu-build.com/> anew, branch "master", just the 7 preselected modules, LFS size "128k".
After flashing the firmware, my ESP8266/4MB says:

```
NodeMCU 3.0.0.0
	branch: master
	commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6
	release: 3.0-master_20190907
	release DTS: 201909070945
	SSL: false
	build type: float
	LFS: 0x0
	modules: file,gpio,net,node,tmr,uart,wifi
 build 2019-09-18 10:26 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
lua: cannot open init.lua
```
So, no LFS. Can you reproduce this behaviour (or PEBKAC on my side)?

marcelstoer, 2019-09-18:
> To be sure, we are talking about the same thing

We are 😉The data in the webform is committed to this repo as `build.config`. A number of scripts then process the configuration and adjust the firmware `.h` files before `make` is invoked to build the firmware.

I'm stretched too thin these days and it'll likely take a while until I can take a look. So, if you can read bash then it's likely faster if you dig into it yourself.

HHHartmann, 2019-09-18:
It is unclear how the defines in build.config relate to the X_ env variables.
There seems to be a script missing.
However it is notable, that in build.config `lfs-size` is defined, while the script checks for `X_LUA_FLASH_STORE` .
should that be `X_LFS_SIZE` instead?

It also does not enter this section at all because `NODEMCU_EAGLEROM_PARTITION` is contained in user_config.h.

IIRC this was an intermediate state, but today the (legacy) setting is honored again. So this check probably should be removed. Not sure about the spiffs settings though.

@marcelstoer Is there any way we can check this?

danjohn, 2019-09-18:
I tried to track down the data flow:

The `web form` posts the value (name="lfsSize") to `trigger-build.php`.
I didn't find the source code of that php script, but hopefully it copies that value to `build.config`, named "lfs-size".
Then the travis script greps for "lfs-size" and exports the value to the bash environment, named "X_LUA_FLASH_STORE".
Last, `set-config.sh` stuffs it into `user_config.h` ("LUA_FLASH_STORE"), from where hopefully `make` picks it up.

What could go wrong?
- `trigger-build.php` doesn't copy "lfsSize" to "lfs-size"
- `set-config.sh` can't do its job bcs of `NODEMCU_EAGLEROM_PARTITION` (as Gregor stated)

`set-config.sh` should spit out a debug message about LFS (to the server log, I suppose; sorry, Marcel, your part :).
I'm not deep enough in the materia to guess what `Eaglerom` is about, but that really could be the Knackpunkt.

@HHHartmann Did you reproduce my example, or just "cold" forensic?
I'd like to rule out, that my setup is causing the problem (someone other should have stumbled over it in the meantime, when the build server delivers ~200 images per day!?)

marcelstoer, 2019-09-18:
@danjohn & @HHHartmann thanks for your feedback. I just postponed hitting the sack for a little longer 😉 Looks like this has been broken for a while (first on `dev` and now also on `master`). Nobody is - seriously - using LFS?

Gregor, you're spot on about `NODEMCU_EAGLEROM_PARTITION`. Here's a recent example; all is public:
- the committed build config https://github.com/marcelstoer/nodemcu-custom-build/commit/403f3def4cfd4068dc624f0291dfac4ad5089341
- the matching Travis CI build log https://travis-ci.org/marcelstoer/nodemcu-custom-build/builds/586706071

That `# Only process LUA_FLASH_STORE and SPIFFS commands if SDK 2.x` at https://github.com/marcelstoer/nodemcu-custom-build/blob/builds/set-config.sh#L16 can now be dumped. We don't need to support SDK 2.x anymore as we have 3.x on `master`. The frozen 1.5 branch can be ignored here as it doesn't support LFS.

danjohn, 2019-09-19:
@marcelstoer If I understood you right, that bug is trapped -- can you estimate, when the changes will be online?

```
NodeMCU 3.0.0.0 
	branch: master
	commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6
	release: 3.0-master_20190907
	release DTS: 201909070945
	SSL: true
	build type: float
	LFS: 0x0
	modules: adc,coap,cron,crypto,dht,file,gpio,http,mdns,net,node,perf,rtcfifo,rtctime,sjson,sntp,tmr,uart,wifi,tls
 build 2019-09-19 05:17 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
lua: cannot open init.lua

```

HHHartmann, 2019-09-19:
@danjohn in the meantime you could use the new commands in the node module to set the partitions for SPIFFS and LFS. 
LFS is always built in now.

danjohn, 2019-09-19:
@HHHartmann Vielen Dank for the kick to `node.[g|s]etpartitiontable()`! I'll have to read thru the changelogs and renewed docs anew, TY!

Sadly, this doesn't fix it.
`node.setpartitiontable{lfs_size=0x20000}` just bricks the chip, it starts up with "l�|�l�b|����r�b�b" (wrong baud rate, obviousely crashes in the xpressiv boot code). The only way to get back is reflashing.

Do you have a working example, how to establish a 128k LFS partition on a ESP8266/4M device with `nodemcu-build.com`-builded firmware? (althru I trust on Marcel to catch the sack^H^H^H^Hbug:)!

HHHartmann, 2019-09-19:
Maybe you need to also set a SPIFFS partition. Unfortunately the documentation does not state what it does exactly.

danjohn, 2019-09-20:
As this seems OT here, I moved the question to `nodemcu-firmware`.

```
	LFS: 0x0
 build 2019-09-20 09:37 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

```
@Marcel Can we help in either way to fix that bug?

marcelstoer, 2019-09-20:
I might find a little time over the weekend to look into this.

> LFS is always built in now.

Then it's all the more surprising the firmware reports `LFS: 0x0`. Maybe those old SDK 2.x instructions actively _disable_ it.

> Can we help in either way to fix that bug?

Yes, create a PR 😜

HHHartmann, 2019-09-20:
@marcelstoer there you go!

closed 2019-09-25 by marcelstoer

marcelstoer, 2019-09-27:
@danjohn the fix is now active.

danjohn, 2019-09-27:
```
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6
	release: 3.0-master_20190907
	release DTS: 201909070945
	SSL: true
	build type: float
	LFS: 0x20000
	modules: adc,coap,cron,crypto,dht,file,gpio,http,mdns,net,node,perf,rtcfifo,rtctime,sjson,sntp,tmr,uart,wifi,tls
 build 2019-09-27 06:20 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
lua: cannot open init.lua

```
Very fine, TY!
You've deserved a breakfast beer! 😜

(Next problem knocks the door: "`Incorrect LFS build type`" when trying to flashreload().
Can this be a side-effect of that fix? I don't think so. Any hints anyway?)

PR #36 (2019-09-25): remove condition to set LFS and SPIFFS configuration in user_config.h
fixes #35 
As detailed in the bug the LFS configuration is also working via user_config.h (legacy) configuration.